Xinyuan-LilyGO/T-Display-S3-AMOLED #4: T-Display-S3 vs AMOLED

dackdel, 2023-05-27:
How different is this from the regular display. I am trying to install https://github.com/BitMaker-hub/NerdMiner_v2 and this was designed for the regular display (here is step by step instruction on how to install: https://www.youtube.com/watch?v=z5GeXZlM4pU)

The problem is the pin out for the screen. Can you help me change the code just for the screen pins?

nspsck, 2023-07-03:
Hi, the problem is actually not only the pin out, changing the pin out is easy as you only have to modify the pins according to the datasheet provided by Lilygo. The bigger problem is that the "TFT_eSPI" library, which the NerdMiner_v2 is based on, doesn't support QSPI configuration. 
After a brief look into the source code, I can suggest you to try the following: 
1. go to: [User_Setup.h](https://github.com/BitMaker-hub/NerdMiner_v2/blob/master/lib/TFT_eSPI/User_Setup.h) and comment out the following from line 169 to 171:
```cpp
#define TFT_CS   PIN_D8  // Chip select control pin D8
#define TFT_DC   PIN_D3  // Data Command control pin
#define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)
```
(Because of the 
```cpp 
#pragma once
``` 
in the pins_config.h later mentioned)

2. in a file named [**User_Setup_Select.h**](https://github.com/BitMaker-hub/NerdMiner_v2/blob/master/lib/TFT_eSPI/User_Setup_Select.h)  comment out the following:
```cpp
#include <User_Setups/Setup206_LilyGo_T_Display_S3.h>
```

3. copy the **pins_config.h, rm67162.cpp and rm67162.h** from [here](https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED/tree/main/examples/factory) to a dictionary you know and include these in the [main program of nerdminer2](https://github.com/BitMaker-hub/NerdMiner_v2/blob/master/src/NerdMinerV2.ino.cpp) and try to compile...

4. fix all the compilation bugs that occurs and upload it to your t-display-s3-amoled. Worst case senarior (most likely to happen), you wasted an afternoon... But if it works, that would be very nice.

bernaction, 2023-07-26:
> Hi, the problem is actually not only the pin out, changing the pin out is easy as you only have to modify the pins according to the datasheet provided by Lilygo. The bigger problem is that the "TFT_eSPI" library, which the NerdMiner_v2 is based on, doesn't support QSPI configuration. After a brief look into the source code, I can suggest you to try the following:
> 
>     1. go to: [User_Setup.h](https://github.com/BitMaker-hub/NerdMiner_v2/blob/master/lib/TFT_eSPI/User_Setup.h) and comment out the following from line 169 to 171:
> 
> 
> ```c++
> #define TFT_CS   PIN_D8  // Chip select control pin D8
> #define TFT_DC   PIN_D3  // Data Command control pin
> #define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)
> ```
> 
> (Because of the
> 
> ```c++
> #pragma once
> ```
> 
> in the pins_config.h later mentioned)
> 
>     2. in a file named [**User_Setup_Select.h**](https://github.com/BitMaker-hub/NerdMiner_v2/blob/master/lib/TFT_eSPI/User_Setup_Select.h)  comment out the following:
> 
> 
> ```c++
> #include <User_Setups/Setup206_LilyGo_T_Display_S3.h>
> ```
> 
>     3. copy the **pins_config.h, rm67162.cpp and rm67162.h** from [here](https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED/tree/main/examples/factory) to a dictionary you know and include these in the [main program of nerdminer2](https://github.com/BitMaker-hub/NerdMiner_v2/blob/master/src/NerdMinerV2.ino.cpp) and try to compile...
> 
>     4. fix all the compilation bugs that occurs and upload it to your t-display-s3-amoled. Worst case senarior (most likely to happen), you wasted an afternoon... But if it works, that would be very nice.

Dont work :(

closed 2024-04-09 by github-actions[bot]