Xinyuan-LilyGO/T-Display-S3-AMOLED #9: SPI instead of QSPI

jpenninkhof, 2023-07-25:
What needs to be done to the board in order to drive it through SPI instead of QSPI? I understand there are soldering pins, but can't find any instructions on what to do with them.

lewisxhe, 2023-07-28:
[image]
Move the two resistors from SPI4 to SPI

lpzsn8888, 2023-07-28:
Does the source code in the example need to be modified?

teastainGit, 2023-07-28:
The supplied example TFT_eSPI_Sprite.ino in the LilyGO github runs fine with QSPI, as shipped.
The board that I received was already QSPI, which is higher performance than plain old SPI.
I imagine that there are other legacy applications that require the older standard SPI.
The resistors shown need to be unsoldered and re-soldered on the SPI option contacts, which is delicate work.

lewisxhe, 2023-07-29:
I have updated my answer and be careful not to damage the screen when welding resistors

jpenninkhof, 2023-07-29:
I was hoping to use it with Tasmota. Tasmota unfortunately doesn’t support QSPI yet and it doesn’t seem to be supported anytime soon. Also, soldering SMD is not something I’m comfortable doing. Hope it can be switched with a jumper or maybe even software on a next version of the board.

closed 2023-07-29 by jpenninkhof