Xinyuan-LilyGO/T-Display-S3-AMOLED #11: lvgl demo example

jrraines, 2023-08-08:
I was able to compile the lvgl demo example and download it. Nothing ever appears on the amoled screen. The serial monitor shows:
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4

assert failed: void setup() lvgl_demo.ino:28 (buf)

Backtrace: 0x4037732a:0x3fcebc90 0x40379ead:0x3fcebcb0 0x4037fe01:0x3fcebcd0 0x42001bda:0x3fcebe00 0x42022bf6:0x3fcebe20

teastainGit, 2023-08-08:
Well, it can be tricky.
It seems that the libraries have changed recently.
To get it to work on Arduino IDE I had to re-clone the GitHub repository and overwrite both the lvgl_demo.ino sketch and the lvgl library with the components from the GitHub repository !
It ran fine last week with the old files!
Other than that, it compiles and runs.
-Terry

jrraines, 2023-08-09:
I cloned it on Aug 6. The sprite demo runs fine. I’m not familiar with the assert syntax but even if I comment that out it crashes with a kernel panic. When I cloned it I downloaded the zip from github. I’m not sure how that actually works—maybe the zip was old? I’ll check that idea out. Thanks for your response!

lewisxhe, 2023-08-09:
Turn on PSRAM and set it to OPI

jrraines, 2023-08-09:
That sounds like the right answer! But I don’t know how to do it. The Readme seems to show a different version of the Arduino tools menu than I have. Readme shows:￼

and I have:
￼

This is Arduino 2.1.1 on a Mac. Poking around in the submenus I don’t see the PSRAM option. Can I set it in setup()?

teastainGit, 2023-08-09:
Some how I cannot see the 'obj' 
[image]

The options are here:
https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED/blob/main/image/Arduino_Config.png

and you enter them in Arduino IDE here:
[image]

-Terry

jrraines, 2023-08-09:
Thank you! now it compiles and runs

jrraines, 2023-11-07:
I started playing with this again. I mostly use the Arduino (2.0+) toolchain. Are there plans to better integrate with that? Proper installation of pins.h in a hardware folder and putting the display into TTF-eSPI’s extensive list of screens in its User folders would greatly improve the compatibility with existing code. I’ve been playing with trying to get Animated Eyes which works on another one of your boards to work and it has been frustrating.

What toolchain are you using?

teastainGit, 2023-11-07:
I am not affiliated with LilyGO!
I use Arduino IDE 2.2.1. 
I keep a copy of 
'User_Setup_Select.h '
handy to use after a Arduino IDE library update.

The board works very well for me in LVGL and TFT_eSPI.

Adapting other applications to this board is tricky.
I tend to keep things simple and I have an example of the LilyGO T-Display S3 AMOLED for TFT_eSPI that you can use as a base, let me know if you want to see it.
-Terry

jrraines, 2023-11-07:
That would be helpful! 

I have seen some of Volos’ videos which use sprites and I’m puzzled why sprites seem to work better (for me as well as Volos) than TFT-eSPI without sprites. The versions of User_Setup_Select.h I’ve seen have not included the rm67162. Animated Eye doesn’t use sprites.

teastainGit, 2023-11-07:
[HotHeadAMOLED.zip](https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED/files/13284317/HotHeadAMOLED.zip)
Note that the rm67162.h and rm67162.cpp are in the sketch folder and the
User_Setup_Select.h tells the TFT_eSPi to use the ```#include <User_Setup.h>           // Default setup is root library folder```
But the <User_Setup.h> ...tells it to use the generic display driver, and the rm67162.h and rm67162.cpp are, as I said, included in your sketch folder.
Very simple, works well.

jrraines, 2023-11-07:
Thank you! I will see if that helps me make progress.

jrraines, 2023-11-10:
This is probably as good as I’m going to struggle for. It’s not right but way better than where I was before.
￼

The good thing about trying is that it finally got my head around the idea that TFT doesn’t actually work at all with the display. I had the misconception that it sort of worked instead of just being a wrapper for sprite which implements a bunch of the TFT commands on a RAM buffer, with the lcd_ commands letting you send the RAM buffer to the screen. I was VERY resistant to seeing that!  Thanks again for your help!

I’ve played with uncanny eyes in a number of forms over the years, using it on a raspberry pi, teensy, MonsterM4sk, etc. 55 years ago, in high school, we spent about 2 weeks looking at polar coordinates. If we’d spent a whole semester on that I’d have probably gotten a failing grade. So I’ve never been able to wrap my head around the geometry involved.

What I suppose would be most helpful for TFT would be to find an implementation with QSPI...

nikthefix, 2023-12-06:
The demo uses a full screen sprite as a frame buffer as TFT_eSPI does not support rm67162 directly. This is a good workaround for any unsupported display as long as you have a separate low level driver which can push raw pixels to the screen. So you compose your graphics by drawing to the sprite frame buffer rather than the usual TFT_eSPI display selection (which is now irrelevant). This opens up loads of options for using newer or more unusual displays with TFT_eSPI - as long as you have the driver, the init codes, the display pixel width & height and plenty of PSRAM.

closed 2024-04-08 by github-actions[bot]