Xinyuan-LilyGO/T-Display-S3-AMOLED #13: Display stopped working

HowdyMoto, 2023-09-16:
Hello,

After receiving my board, I left it plugged in for a couple days and then woke up and the display no longer works.

When I plug the device in, the red charging LED workd.
After reloading the firmware.bin, the device shows the red charging LED, and the green LED on the top of the board flashes repeatedly, but the display never lights up or shows anything. The device shows up in Arduino IDE and I'm able to successfully uploaded sketches to it.

Is the display broken or are there some ways to try to get it working again?

asiorgad, 2023-11-28:
Have the same issue , they are selling crap .

nikthefix, 2023-12-06:
My T-Track Amoled died in the same way. I think it's ESD damage. Take care with grounding your workspace and body when handling these boards - all the usual antistatic precautions apply.

asiorgad, 2023-12-06:
For me , at the end ,the board was actually ok , I just flashed the regular "T-Display S3" firmware instead of the "T-Display S3 AMOLED"
So just make sure you're flashing the right one

asiorgad, 2023-12-06:
Their technical support is awful ..
That's my experience...
even though I managed to fix it at the end , when I was facing issues with the board - my experience with the support team was really bad, and I probably won't purchase any devices from them again .

asiorgad, 2023-12-06:
That's right, and at first sight they seemed to be a company with qualified products, but actually their have lack of tech support, warranty, and customer service.

teastainGit, 2023-12-06:
Hi, everyone! I live in Canada and I do not work for LilyGO !!!
I own two LilyGO T-Display S3, one LilyGO T-Display S3 AMOLED and a T-HMI.
They all work splendidly, I can run all the examples and I've created my own test programs called HotHead for each board.
My first LilyGO T-Display S3 was a nightmare to get running because:
The demo program is useless until you edit it with your own WiFi creds.
Using other peoples sketches for other ESP32s is frustrating because you must add power enable and back light outputs for it to work. (at-all in some cases).
To support other users with the LilyGO T-Display S3 I made this GitHub repository:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 
The AMOLED and T-HMI are easier to use, so I did not make a repository, but my HotHead test programs are available on request.
-Terry

nikthefix, 2023-12-07:
Lilygo are unusual in being so prolific and having so many interesting ideas which they quickly offer up to the maker community in the form of affordable hardware. I can't think of another firm as enthusiastically pumping out unique new boards. Tech support is not something I would expect from the sole maintainer of the whole lilygo codebase. Impossible job. But community support is good and these github pages are very active. Espressif support is also really good. New boards will obviously take time for the collective user knowledge base to build.

@Terry, Thanks for sharing your work.

julio-matarranz, 2024-02-06:
Mine died yesterday, I tried factory firmware but there is no back light anymore. I don't know what happened it was just USB powered from the computer because I was trying to code something about elm327 wifi based on nerdminer code.
I haven't been using this board much (probably 3h)... Any idea ?

[image]

nikthefix, 2024-02-06:
@julio-matarranz 
The AMOLED doesn't have a backlight. Each pixel generates it's own light. From your picture it looks like you still have some visible activity on the screen. Does your sketch display as expected albeit very dimly?

If so then I suspect the AMOLED power management IC has failed.
Same thing happened to me with the T-Track-AMOLED. I couldn't bring it back to life but I concluded that I needed to improve my anti-static handling of the board. 

Another possibility it that the display brightness register retains a low value. Try sending 0xFF to the brightness control register to see if it has any effect.

nik

julio-matarranz, 2024-02-06:
> @julio-matarranz The AMOLED doesn't have a backlight. Each pixel generates it's own light. From your picture it looks like you still have some visible activity on the screen. Does your sketch display as expected albeit very dimly?
> 
> If so then I suspect the AMOLED power management IC has failed. Same thing happened to me with the T-Track-AMOLED. I couldn't bring it back to life but I concluded that I needed to improve my anti-static handling of the board.
> 
> Another possibility it that the display brightness register retains a low value. Try sending 0xFF to the brightness control register to see if it has any effect.
> 
> These setting should be volatile but I've found that sometimes they hold thru a power cycle unless they are explicitly initialized. Not sure why. Try adding delays in setup for a decent power on reset.
> 
> Additional: Did it fail after an upload or just unexpectedly when you plugged it in? The reason I ask is that updates to the lilygo library to include the touch version and Mk2 version of this display have a 'display enable' on pin 38 (I think it's 38 but I'm away from my dev machine at the moment) which needs to be pulled high to enable the display. Did you change your libs in any way before the failure?
> 
> nik

Dear, thank you for your answer it's really usefull to understand how AMOLED works. My picture is with original firmware (downloaded the oldest bin from older commit on the repository). I don't know how I broke it but I have 3d printed a case so the board was protected... Only plugged USB on laptop to work on the custom firmware but the screen wasn't working anymore.

julio-matarranz, 2024-02-08:
Dear, based on octagon_on_amoled I produced this simple code : 

```
#include "rm67162.h"
#include <TFT_eSPI.h>

// Initialize display and sprites
TFT_eSPI tft = TFT_eSPI();
TFT_eSprite bgr = TFT_eSprite(&tft);
TFT_eSprite sprite = TFT_eSprite(&tft);

void setup()
{
    Serial.begin(115200);
    sleep(10);
    Serial.println("T-DISPLAY-S3-AMOLED SCREEN TEST");
    rm67162_init();
    lcd_setRotation(1);
    sprite.createSprite(536, 240);
    sprite.setSwapBytes(1);
}

void loop()
{
    Serial.println("lcd_on");
    lcd_on();
    for (uint16_t color : default_4bit_palette)
    {
        sprite.fillScreen(color);
        lcd_PushColors(0, 0, 536, 240, (uint16_t *)sprite.getPointer());
        Serial.print("fill ");
        Serial.println(color, HEX);
        sleep(5);
    }
    sprite.fillScreen(0xFF);
    lcd_PushColors(0, 0, 536, 240, (uint16_t *)sprite.getPointer());
    Serial.println("fill 0xFF");
    sleep(5);
    Serial.println("lcd_off");
    lcd_off();
    sleep(5);
}
```

The screen keep staying dark. Do you have other advice ?

julio-matarranz, 2024-02-08:
I took the master from this project on platformio
put src_dir = examples/octagon_on_amoled in platformio.ini and tweeked the "octagon_on_amoled.ino" file.
I found out that my screen is "working" if I manage less pixels. I suppose that my screen isn't powered enough anymore.

```
// Include necessary libraries and fonts
#include "rm67162.h"
#include <TFT_eSPI.h>

// Initialize display and sprites
TFT_eSPI tft = TFT_eSPI();
TFT_eSprite bgr = TFT_eSprite(&tft);
TFT_eSprite sprite = TFT_eSprite(&tft);

//#define LCD_H_RES     536
//#define LCD_V_RES     240

// #define LCD_H_START     0
// #define LCD_V_START     0

#define LCD_H_RES     50
#define LCD_V_RES     50

#define LCD_H_START     100
#define LCD_V_START     100

void setup()
{
    Serial.begin(115200);
    sleep(10);
    Serial.println("T-DISPLAY-S3-AMOLED SCREEN TEST");
    rm67162_init();
    lcd_setRotation(1);
    sprite.createSprite(LCD_H_RES, LCD_V_RES);
    sprite.setSwapBytes(1);
}

void loop()
{
    sprite.fillScreen(TFT_BLACK);
    lcd_PushColors(LCD_H_START, LCD_V_START, LCD_H_RES, LCD_V_RES, (uint16_t *)sprite.getPointer());
    Serial.println("fill TFT_BLACK");
    sleep(2);
    sprite.fillScreen(TFT_WHITE);
    lcd_PushColors(LCD_H_START, LCD_V_START, LCD_H_RES, LCD_V_RES, (uint16_t *)sprite.getPointer());
    Serial.println("fill TFT_WHITE");
    sleep(2);
    sprite.fillScreen(TFT_GREEN);
    lcd_PushColors(LCD_H_START, LCD_V_START, LCD_H_RES, LCD_V_RES, (uint16_t *)sprite.getPointer());
    Serial.println("fill TFT_GREEN");
    sleep(2);
    sprite.fillScreen(TFT_RED);
    lcd_PushColors(LCD_H_START, LCD_V_START, LCD_H_RES, LCD_V_RES, (uint16_t *)sprite.getPointer());
    Serial.println("fill TFT_RED");
    sleep(2);
    sprite.fillScreen(TFT_BLUE);
    lcd_PushColors(LCD_H_START, LCD_V_START, LCD_H_RES, LCD_V_RES, (uint16_t *)sprite.getPointer());
    Serial.println("fill TFT_RED");
    sleep(2);
}
```

nikthefix, 2024-02-08:
I see. Well unfortunately that does suggest the power management IC has failed. This IC provides the -ve and +ve supply required by the display. I bet the vendor will send you another if you explain what happened.

julio-matarranz, 2024-02-08:
> I see. Well unfortunately that does suggest the power management IC has failed. This IC provides the -ve and +ve supply required by the display. I bet the vendor will send you another if you explain what happened.

Thank you for your help, I wish I could fix or replace by my own but I guess it's complicated.
I asked for support already but they mentionned "spring festival holiday" so I have to wait until 21st feb to get an answer. :(

Thank you again ;)

nikthefix, 2024-02-08:
Yes. Everything stops for a month. 

The IC is replaceable and readily available but it's very small. I wouldn't attempt to replace it myself for fear of damaging the other components on the board.

There is a manufacturer producing a version of this display (and others used by Lilygo) which has the power management IC built into the FFC ribbon cable with additional ESD protection. So if you ever want to make your own board then these could be a good option. I've been experimenting with a few of these and they're really good.

Link: https://www.dwo.net.cn/

nik

3Dsf-info, 2024-02-08:
> I wonder how bad it is to remove the screen from the board (it seems to be hardly glued), I will wait seller's answer first.

The AMOLED display is removable from the Lilygo T-Display S3 AMOLED board, yes. But the ZIF connectors and the ribbon cable are extremely fragile and easily damaged, so I wouldn’t recommend it particularly unless you have to.

 In my case I’m offsetting the display at a higher distance from the board, so I’m taking the chance. And I’ve bought some spare displays…

 - NK Guy

nikthefix, 2024-02-09:
@3Dsf-info 

It's good that the bare displays are so affordable. I've experimented with many similar sized AMOLEDs and don't find them particularly fragile - I've not destroyed one even tho I've swapped them between boards (ZIF and soldered) and often put quite a bit of strain on the FFC /FPC. There's also a lot more flex than with an IPS-TFT.

The failure always seems to be the power controller IC but I suspect that this may be due to a combination of BGA footprint and poor reflow soldering on my part.

Nevertheless, I've always found charge pumped sources to be vulnerable. I'm forever losing DC-DC converters.

nik

nikthefix, 2024-02-09:
Indeed! Good point.
Plus, and speaking generally, there's often so much glue!

nikthefix, 2024-02-09:
Here's a clip of one of the DWO amoled displays mentioned above.
It's 360x360 and has power management built in so you only need to worry about QSPI and 3.3v power.

[dwo_amoled.zip](https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED/files/14220460/dwo_amoled.zip)

closed 2024-04-08 by github-actions[bot]