Xinyuan-LilyGO/T-Display-S3-AMOLED #19: What board to select for T-Display-S3-AMOLED

martinglass, 2023-12-15:
What board to select for the T-Display-S3-AMOLED?

I've followed the README.md file in this repo, but there appears to be a crucial step missing in the instructions for the Arduino IDE.

Step 5 says "Select the correct settings", but there are no instructions beforehand about what board to select. The obvious choice T-Display-S3-Amoled does not show up in the list of ESP32 boards.

_**4. Copy everything from this repository lib folder to [Arduino library folder](https://docs.arduino.cc/software/ide-v1/tutorials/installing-libraries#manual-installation) (e.g. C:\Users\YourName\Documents\Arduino\libraries)
5 .Select the correct settings in the Tools menu as shown below.**_

Can someone help, please?

PS I've noticed that in this repo's board folder there is a json file for the s3 Amoled board, but it's not showing up in the Arduino IDE.
https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED/blob/main/boards/lilygo-t-amoled.json

3Dsf-info, 2023-12-15:
I found getting information on the board quite difficult. I wrote up some stuff on what worked for me. Halfway down the page.

 https://sites.google.com/site/3dsfinfo/lights-for-model-makers-technology-and-techniques/high-rez-microcontroller-displays

 - NK Guy

martinglass, 2023-12-15:
@3Dsf-info many thanks. In short, you say the board to select is "ESP32S3 Dev Module"

PS I loved your page. Great stuff.

PPS If you've got the money, how about this listing I just saw on Facebook : https://www.facebook.com/marketplace/item/671111165151627/

3Dsf-info, 2023-12-15:
All I can say is that it worked for me.

 Thanks for the kind words. I’ll pass on the droid replica, though. It was made by a certain notorious builder and isn’t screen-accurate. :)

 - NK Guy

lewisxhe, 2023-12-16:
ESP32 S3 Dev Module

martinglass, 2023-12-16:
Hi Lewis,

Thanks for confirming that.

Not sure this is the best place to give feedback, but I just want to say a big thank you for all the LilyGo products you do. They're really great and getting better all the time.

Regards,
Martin

lewisxhe, 2023-12-16:
This repository was created by @mmMicky earlier, [LilyGo-AMOLED-Series](https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series) also supports the non-touch version

closed 2024-04-07 by github-actions[bot]