Xinyuan-LilyGO/T-Display-S3-AMOLED #20: Examples here do not work with latest version of LVGL library (currently version 8.3.11)

martinglass, 2023-12-16:
The examples given here do not work with the current release of LVGL, version 8.3.11

As a workaround, I had to delete the Arduino IDE installed lvgl folder from my Arduino/libraries/lvgl folder and replace it with the one here, in the repo folder lib/lvgl. The one in this repo is based on 8.3.0-dev.

Can this be corrected, please, because it causes unnecessary problems for users?

lewisxhe, 2023-12-16:
Out of service? Or a compilation error?

lewisxhe, 2023-12-16:
```
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Converting factory.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- lvgl @ 8.3.11
|-- OneButton @ 2.0.3
|-- WiFi @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- SPI @ 2.0.0
Building in release mode
Compiling .pio\build\T-Display-AMOLED\src\factory.ino.cpp.o
Compiling .pio\build\T-Display-AMOLED\src\factory_gui.cpp.o
Compiling .pio\build\T-Display-AMOLED\src\font_Alibaba.c.o
Compiling .pio\build\T-Display-AMOLED\src\lilygo1_gif.c.o
Compiling .pio\build\T-Display-AMOLED\src\rm67162.cpp.o
Building .pio\build\T-Display-AMOLED\bootloader.bin
Generating partitions .pio\build\T-Display-AMOLED\partitions.bin
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\core\lv_disp.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\core\lv_event.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\core\lv_group.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\core\lv_indev.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\core\lv_indev_scroll.c.o
esptool.py v4.5.1
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\core\lv_obj.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\core\lv_obj_class.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\core\lv_obj_draw.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\core\lv_obj_pos.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\core\lv_obj_scroll.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\core\lv_obj_style.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\core\lv_obj_style_gen.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\core\lv_obj_tree.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\core\lv_refr.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\core\lv_theme.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\arm2d\lv_gpu_arm2d.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\lv_draw.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\lv_draw_arc.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\lv_draw_img.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\lv_draw_label.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\lv_draw_layer.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\lv_draw_line.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\lv_draw_mask.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\lv_draw_rect.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\lv_draw_transform.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\lv_draw_triangle.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\lv_img_buf.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\lv_img_cache.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\lv_img_decoder.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\nxp\pxp\lv_draw_pxp.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\nxp\pxp\lv_draw_pxp_blend.c.o
C:/Users/lewis/Desktop/T-Display-S3-AMOLED/examples/factory/factory.ino:16:2: warning: #warning "If you need to monitor printed data, be sure to set USB CDC On boot to ENABLE, otherwise you will not see any data in the serial monitor" [-Wcpp]
 #warning "If you need to monitor printed data, be sure to set USB CDC On boot to ENABLE, otherwise you will not see 
any data in the serial monitor"
  ^~~~~~~
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\nxp\pxp\lv_gpu_nxp_pxp.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\nxp\pxp\lv_gpu_nxp_pxp_osa.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\nxp\vglite\lv_draw_vglite.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\nxp\vglite\lv_draw_vglite_arc.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\nxp\vglite\lv_draw_vglite_blend.c.o
Compiling .pio\build\T-Display-AMOLED\lib750\lvgl\draw\nxp\vglite\lv_draw_vglite_line.c.o
[... 214 lines cut ...]
Compiling .pio\build\T-Display-AMOLED\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\T-Display-AMOLED\FrameworkArduino\firmware_msc_fat.c.o
Compiling .pio\build\T-Display-AMOLED\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\T-Display-AMOLED\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\T-Display-AMOLED\FrameworkArduino\main.cpp.o
Compiling .pio\build\T-Display-AMOLED\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\T-Display-AMOLED\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\T-Display-AMOLED\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\T-Display-AMOLED\libFrameworkArduino.a
Linking .pio\build\T-Display-AMOLED\firmware.elf
Retrieving maximum program size .pio\build\T-Display-AMOLED\firmware.elf
Checking size .pio\build\T-Display-AMOLED\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  19.3% (used 63280 bytes from 327680 bytes)
Flash: [==        ]  19.2% (used 1259393 bytes from 6553600 bytes)
Building .pio\build\T-Display-AMOLED\firmware.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 3 ELF sections
Successfully created esp32s3 image.
=========================================== [SUCCESS] Took 30.26 seconds ===========================================
 *  Terminal will be reused by tasks, press any key to close it. 
```

Compiling on platformio doesn't show any errors, since I don't have the board I can't verify that it runs correctly

martinglass, 2023-12-16:
> Out of service? Or a compilation error?

Compilation error. lv.h not found in expected location etc

lewisxhe, 2023-12-16:
You should place the lv_conf.h in the warehouse in the C:\Users\YourName\Documents\Arduino\libraries directory. Is this correct? Upgrading only requires replacing the lvgl folder, and lv_conf.h should remain in the original directory.

If it still doesn’t work, please take a screenshot and tell me what files are contained in the Arduino\libraries directory.

closed 2024-04-07 by github-actions[bot]