Xinyuan-LilyGO/T-Display-S3-AMOLED #21: How to break reboot loop to allow for flashing a new image?

mavnorthwind, 2024-01-04:
I've played around with a LilyGO S3 AMOLED board and after successfully flashing a modified "Factory" sample, I seem to have uploaded a different sketch from the Examples folder that is unsuitable for my board. Now the ESP32 is hanging in a reboot loop and I cannot upload a new image.

When the board is connected to my Win10 machine, the serial monitor starts repeating
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420581e2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0

assert failed: virtual void LilyGo_AMOLED::pushColors(uint16_t, uint16_t, uint16_t, uint16_t, uint16_t*) c:\Users\MyUserName\Documents\Arduino\libraries\LilyGo-AMOLED-Series\src\LilyGo_AMOLED.cpp:649 (pBuff

Backtrace: 0x40378012:0x3fcebb20 0x4037b885:0x3fcebb40 0x40381c5d:0x3fcebb60 0x42001e7a:0x3fcebc90 0x42001cd4:0x3fcebcc0 0x42055d71:0x3fcebce0 0x4200d631:0x3fcebd10 0x4200de4a:0x3fcebde0 0x42025fac:0x3fcebe60 0x42001af3:0x3fcebe90 0x42030c51:0x3fcebeb0

ELF file SHA256: eb9b7599e70729a4

Rebooting...
```
over and over again.
Trying to upload the "Factory" sketch again via the Arduino IDE fails with `Failed chip erase: uploading error: exit status 2`.
I guess because the arduino reboots while the IDE still tries to setup upload parameters?

I hope I didn't brick my device, so how can I get the board programmable again?

I'm still a beginner, so please describe in detail what to do.

Thanks in advance!

myspac0, 2024-01-04:
- verify options in tools menu with lilygo details for this module
-  unplug the module
- press reset and keep button
- plug the module
- wait some 3 - 5 second
- unpress reset

You coud upload your code (with right settings !!!)

mavnorthwind, 2024-01-05:
Thanks, I've found out how to do it and now have the factory firmware running again.
For reference (if anyone else has a similar problem), I'll try explaining the steps in detail here:
- The ESP32 can be put in flashing mode, where current firmware is not being executed during boot. In this mode, the ESP32 is only waiting for a new firmware to be flashed.
- In order to put the ESP32 into flashing mode, you have to have GPIO Pin 0 connected to GND while rebooting.
- The problem was that I could not find GPIO0 on the backside of the LilyGO S3 AMOLED board.
- There's an [image in this repo](https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED/blob/main/image/T-Display-S3-AMOLED.jpg) showing the pin layout and so I was able to find that GPIO0 seems to be connected to one of the buttons next to the USB connector (labelled `(IO0)Boot` in the image).
- So now I connected the S3 AMOLED to my machine via USB and kept the "Boot" button pressed while pressing the reset button.
- I held the Boot button for about 10 seconds before looking at the serial port output again (using `type COM8:` in a windows shell for simplicity).
- Indeed, the serial port didn't output anything, so the reboot cycle seemed to have been broken.
- Finally I was able to upload the factory firmware to the board using the Arduino IDE again.

While this may be obvious for the pros out there, fixing such a situation as a beginner can be really challenging.

hth
mav

closed 2024-01-05 by mavnorthwind

teastainGit, 2024-01-05:
Hey mav, nicely written, glad you got it working.
They are 'un-brickable' in that the bootloader is in hard wired ROM.
BTW, I've learned to Never compile the Arduino IDE 'NEW Sketch' without adding 100ms delay in the Loop!
And if your test program won't re-start...un-plug the USB wait a second and plug back in.
I keep a minimum sketch with serial print tick/tock for diagnostic.
Cheers Terry