Xinyuan-LilyGO/T-Display-S3-AMOLED #23: device not responding, no COM port assignment

TobiasPSP, 2024-01-15:
I am experienced in ESP8266 but new to AMOLED and ESP32. The AMOLED T-Display initially showed the preinstalled software, so display and device worked.

In an effort to get a first own working code example running, apparently I accidentally uploaded code targeting an LCD version of T-Display, not AMOLED.  

After uploading succeeded, the device is now powering up when I connect it to USB, and the Windows 10 device manager also emits a sound on connect, however instead of getting a COM port assigned, in device manager I just see an "Unknown USB Device" (device descriptor request failed), and neither platformio nor your firmware update tool seem to be able to connect to the device anymore.

What can I do to get the device back in a responsive state? Many thanks.

lewisxhe, 2024-01-15:
Press and hold the BOOT button, don't release it, then insert the USBC, and finally release the BOOT button

TobiasPSP, 2024-01-15:
Many thanks, now I have a COM port. 
Next I installed the ESP32S3 flash update tool V3.9.5 and tried to flash any of the bin files found here: https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED/tree/main/firmware. 
I selected one of the bin files and added a "0" after the "@". Next, I used these options:
40Mhz, QIO, DoNotChgBin checked, COMport, 921600 baud.
When I click START, the icon in the tool changes to a "sync", then a green "download", then a blue "finish".
Next to finish, I see this:
AP:  348518A9948D  STA:  348518A9948C
BT:  348518A9948E  ETHERNET:  348518A9948F

After flashing, when I normally reset the device, all is as before. No COM port, blank screen. 
I was under the impression when I flash i.e. "[T-Display-AMOLED_Cam_V1.1.bin" that this included the basic firmware plus the default software, so I expected to see the same display messages as OOB.

What would I be doing wrong? Many thanks for any suggestion.

TobiasPSP, 2024-01-15:
My suspicion is that the firmware wasn't updated somehow. The firmware update tool does not show any error message though. I was surprised to see the firmware update take only 2-3 seconds. Given the size of the update at 1,3MB, even at 921.600 baud I would expect the upload to take around 15 seconds. The tool however changes from "downloading" to "finished" within 2-3 seconds.

TobiasPSP, 2024-01-15:
OMG. Problem solved. I missed to set the checkmark before the line with the loaded bin file. 
Sorry for that, now all works as expected.

TobiasPSP, 2024-01-15:
Maybe in a future flashtool version, when there are no bin files selected and the user still clicks "start", a warning would be in order ;-). I know it is all my fault but any bit of added user friendlyness helps all sides...

closed 2024-04-07 by github-actions[bot]