Xinyuan-LilyGO/T-Display-S3-AMOLED #24: SD Card Reader Module with AMOLED Touch Version using SPI

EvanZhou1999, 2024-01-18:
Hi, I'm new to ESP32S3. I want to connect an sd card reader module (with standard SPI Interface cs/mosi/clk/miso) to the AMOLED touch version. Although it looks like the screen is already using the QSPI bus, I wonder if there are other SPI pins available on the board? (I see FSPI pins on the diagram but don't know how to use them). Can anyone point me to the right direction? Any help is appreciated!

lewisxhe, 2024-01-19:
As long as there is no occupied GPIO, you can use it. You only need to specify the SPI Pin before initialization. Here is a short example.
```
const int sck = 1;
const int miso = 2;
const int mosi = 3;
const int cs = 10;

SPI.begin(sck, miso, mosi);
// Set mount point to /fs
if (!SD.begin(cs)) {
    Serial.println("Failed to dected SDCard!");
}
if (SD.cardType() != CARD_NONE) {
    Serial.printf("SD Card Size: %llu MB\n", SD.cardSize() / (1024 * 1024));
}
```

EvanZhou1999, 2024-01-28:
Hi, Thank you so much for the help! with a little bit of extra research I found that the FSPI bus on the s3 works for attaching SPI peripherals, just putting it out there for anyone who is looking for an answer:

Pin 13 = FSPIQ = MISO
Pin 12 = FSPICLK = CLK
Pin 11 = FSPID = MOSI
Pin 10 = FSPICS0 = CS

The SPI can be declared like this:
```
SPIClass sdSPI(FSPI);
```
And then in the setup function:
```
sdSPI.begin(12, 13, 11, 10);
if(!SD.begin(10, sdSPI)){   // cs pin
    Serial.println("Card Mount Failed");
    return;
}
// more codes to mount the files….
```

closed 2024-01-28 by EvanZhou1999