Xinyuan-LilyGO/T-Display-S3-AMOLED #25: T-Display-S3-AMOLED stop showing anything on the screen

BMEWCL2, 2024-02-19:
When I powered up my device this morning, the screen went blank. Is there a way to fix this issue?

I tried different demo programs, but none of them worked. Is it a hardware issue?  If so, would it be possible to confirm it?  

Please let me know your feedback and suggestions.

Best

Will

teastainGit, 2024-02-19:
What IDE are you using?
It may have done a Library update and overwritten the libraries  from the LilyGO T-Display S3 AMOLED  repository .
Try copying all the repository libraries back into your local library . 
Choose a simple sketch to see if that is the issue.
I have a LilyGO T-Display S3 AMOLED test sketch for TFT_eSPI  that has low requirements!
Let me know, Terry

julio-matarranz, 2024-02-19:
Dear,
You can check this issue : https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED/issues/13#issuecomment-1934407176
Good luck 🙏🏻

closed 2024-04-07 by github-actions[bot]