Xinyuan-LilyGO/T-Display-S3-AMOLED #32: Board Serial problem

08822407d, 2024-04-25:
My board does not send message(or to say after start monitoring the port) when running both factory test and my program. I program with platformio extension for vscode, build with arduino framework.
Is there any bug, or need I do some configure?

The reason I suppose it does not send message is that, when I monitoring in auto connect mode, the monitor can recieve os boot message like`ELF file SHA256: 76ef6c145979946f

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420cdafa
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[ 31482][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -1
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400556d5  PS      : 0x00060130  A0      : 0x820b9e6e  A1      : 0x3fcebca0  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3fcebc60  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00060d20  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000003  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  

Backtrace: 0x400556d2:0x3fcebca0 |<-CORRUPTED`

lewisxhe, 2024-04-25:
This is caused by the expiration of the ipapi.co website certificate, which can be solved by updating the certificate.

lewisxhe, 2024-04-25:
The certificate has been updated and there are no problems when I test it.

closed 2024-04-25 by lewisxhe

08822407d, 2024-04-26:
> This is caused by the expiration of the ipapi.co website certificate, which can be solved by updating the certificate.
Would you please tell about how to update the certificate or where can I find a valid guide? I'm greenhand on embedded developing.

08822407d, 2024-04-26:
> This is caused by the expiration of the ipapi.co website certificate, which can be solved by updating the certificate.

I update the repo and tested, Serial Monitor still doesn't recieve any message from board. I read exmaple code carefully and confirm that before getting date from time server, it have send some messages through serial, for example the result of Wifi-Scan.

crosser, 2024-08-07:
As of today, factory/example fails to connect to ipapi.co (with misleading error message "connection refused") because of wrong CA ceritificates, and reboots. The site's certificate is (currently) signed by **C = US, O = Google Trust Services LLC, CN = GTS Root R4**. Adding their certificate to the bundle in `zones.h` fixes the problem. CA certificate can be found here https://i.pki.goog/r4.pem.