Xinyuan-LilyGO/T-Display-S3-AMOLED #33: Connecting WS2812 5050 LED Strip

JpEncausse, 2024-05-21:
Hello,
I was wondering if it is possible to connect WS2812 5050 LED Strip with FastLED? 
The strip require 5V if I understood.

I have both old and new version. According to the scema it is written
- Old version  VBus VSYS
- New version : VBus +5V
I was wondering what is the difference ? What VSYS mean ? is it 5V also ?

I tested on Old version it doesn"t work. If I plug on 3.3V it works but LED are all scramble. Not enought power I think
Could you provide some clarification ?

Thanks !

teastainGit, 2024-05-22:
A large strip cannot practically powered by the board, it requires a separate power supply and just join the grounds together.

JpEncausse, 2024-05-22:
The fun fact is that it works on a small TTGO-T-Display but not on the AMOLED
Do I have to "power" both GND ? (I leave one empty)

JpEncausse, 2024-05-23:
My bad, it seems to works with 3.3V. This weird behavior seems to be related with a bug in the code that was replicated on 2 test devices. So everythings fine !

closed 2024-05-23 by JpEncausse