Xinyuan-LilyGO/T-Display-S3-AMOLED #36: External Wifi Antenna

yt38, 2024-08-29:
Hi, I am having trouble with wifi range. I plugged in an external antenna but it does not seem to make much difference. Do I have to make any mods in order to use the external antenna instead of the internal one?

Thank you

sketchysunday, 2024-08-30:
Just like the non-AMOLED T-Dsiplay-S3, you have to desolder the 0ohm resistor that goes to the onboard antenna and turn it 90 degrees to connect it to the trace for the external antenna connector.

Myself, I don't turn the resistor, I just remove it and connect the traces with solder.
Don't expect more than +5dB with a FPC antenna but results may vary.

Here;s a picture of the antenna section of non-AMOLED display, it looks similar:

[image]

Ref: https://github.com/Xinyuan-LilyGO/T-Display-S3/issues/188

yt38, 2024-09-02:
Thank you for the reply. I might try it. I have both the non-touch V1 and touch V2 and found that touch V2 has better ESPNOW broadcast performance (internal antenna). I haven't measured the exact signal strength but distance wise, it out performs V1 by a lot. For now I will use the V2 as master and V1 as slave.

closed 2024-09-02 by yt38