Xinyuan-LilyGO/T-Display-S3-AMOLED #37: 3D Model for the T-Display-S3-AMOLED V2 (No Touch)

aaronbeckmann, 2025-02-28:
### Hello everyone,

I have seen that this repository includes a 3D model (.step file) for the T-Display-S3-AMOLED with touch functionality (60mm length) at 3D_File/t-display-s3-amoled.stp. 

Could you add the model for second version (55mm length) of the non touch T-Display-S3-AMOLED also? This would make including the amazing board into custom projects much easier. 

Thanks a lot!

lewisxhe, 2025-02-28:
Is this what you are talking about ?
https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED/tree/main/3D_File

aaronbeckmann, 2025-02-28:
Yes, this step file represents the T-Display-S3-AMOLED with touch (which is 60mm long):

[image]

The second version of the  T-Display-S3-AMOLED is now 55mm long (https://lilygo.cc/products/t-display-s3-amoled?variant=43271614267573)

lewisxhe, 2025-02-28:
T-Display-S3-AMOLED non-touch version? The length has changed? Can you take a screenshot and tell me which version it is? Because there are too many

lewisxhe, 2025-02-28:
Well, this board is not in this repository, it is here https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/tree/master/shell

aaronbeckmann, 2025-02-28:
The .step file for the T-Display-S3-AMOLED in this repository (https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/master/shell/T-Display-S3-AMOLED.zip) looks like this:

[image]

Just a housing without any PCB.

aaronbeckmann, 2025-02-28:
This file: https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/master/dimensions/T-Display-S3-AMOLED-Touch-PCB-3D.7z contains a model a touch version with 55mm length. The layout of the components of the PCB look exactly like _Touch Non-Soldered Pin [H681]_. The PCB itself looks like its the same as _V2.0 Non-Soldered Pin [H713]_ just with another display and acrylic mounting piece. 

[image]

This file: https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/master/dimensions/T-Display-S3-AMOLED-Plus.7z contains a model without display or acrylic mounting piece which features a micro SD card slot. A version like shown there is not being sold:

[image]

While the first model has the correct PCB there are differences in the display and in the acrylic mounting piece. I think I could still work with that but if you had an actual model this would be great.

lewisxhe, 2025-02-28:
No! Because this model is the touch version without the touch panel.

aaronbeckmann, 2025-02-28:
Okay, then I'll be using this model: https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/blob/master/dimensions/T-Display-S3-AMOLED-Touch-PCB-3D.7z and do some minor adjustments myself. 

Thanks a lot for the help and effort!

closed 2025-02-28 by aaronbeckmann