Xinyuan-LilyGO/T-Display-S3-AMOLED #38: Setup of CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN for more BLE devices

JpEncausse, 2025-03-02:
Hello,
It seems the board is limited to 4 BLE device connection. ESP-32 should be able to handle up to 7.
To do that it seems I have to do `CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN=7`
But it seems it doesn't work when I put the `sdkconfig.defaults` aside of `platformio.ini file`.
Any advice ?

lewisxhe, 2025-03-03:
Are you using idf or arduino?

arduino cannot be modified, this is fixed when compiling the sdk, and the CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN macro is invalid on esp32S3.

This has nothing to do with the board, because the difference in hardware peripherals does not affect the wireless function. Please try using the idf example

JpEncausse, 2025-03-03:
Thanks I'm using platformio with this repository. I'll look into idf

closed 2025-04-17 by github-actions[bot]