Xinyuan-LilyGO/T-Display-S3-AMOLED #39: Dead screen

jernejknechtl, 2025-04-26:
Hi,

I have two Lilygo T-Display S3 AMOLED boards. One of them seems to have a dead screen. I was using it occasionally (couple of min every other day) as a display for a Bluetooth device, and everything was working fine until the display started showing signs of failure. It gradually became dimmer, the colours looked washed out, and eventually, it stopped displaying anything at all.

Is there anything I can try to "fix" the board, or is it completely dead?
The other one is still working for now, but it's a bit disappointing to see this happen on a device that isn’t exactly cheap.

3Dsf-info, 2025-04-26:
The AMOLED display is a DO0200FS01 made by DWO. You could buy a replacement display and see if that fixes it. Swapping the screen out is tricky since the connectors and cables are so fragile, and the original display is taped to a clear plastic frame. But it is doable.

https://sites.google.com/site/3dsfinfo/lights-for-model-makers-technology-and-techniques/animated-gifs-and-the-lilygo-t-display-s3-amoled

https://www.dwo.net.cn/pd.jsp?id=11919&fromColId=0#pfc=%7B%7D&_pp=0_316_1_-1

lewisxhe, 2025-04-28:
If you are sure that the screen is damaged, you can contact the dealer and they will negotiate with you.

jernejknechtl, 2025-05-04:
Now even the screen on the second one is dead. I was running the same code on them for months and all was fine. Could the issue be somehow related to running the board on 3.7V Lipo Battery 620mAh ?   
The second one was fine on cable but stopped working while on battery.

lewisxhe, 2025-05-05:
When connected to a battery for power supply, is there any voltage output from the 3.3V pins on both sides of the board?

closed 2025-06-19 by github-actions[bot]