Xinyuan-LilyGO/T-Display-S3-AMOLED #40: How best to power with external power

sharknoms, 2025-05-05:
Hi, could you please advice on how to best power with external power.
Can i use a 5v external power supply on the BAT connector?

Thank you.

lewisxhe, 2025-05-06:
BAT is the battery connection, the maximum is 4.2V, if you use an external power supply, you can connect it from VBUSPin. The maximum input is 5V

sharknoms, 2025-05-06:
Thanks, so the issue described here is not a problem for the AMOLED display?  https://github.com/Xinyuan-LilyGO/T-Display-S3/issues/205  "USB 5V is available on one external pin as an output only to supply some sensors with 5V, but cannot be used to backfeed 5V to the board."

lewisxhe, 2025-05-06:
That is the board, not the AMOLED. The AMOLED does not have a diode to control the direction and can be powered by an external power supply.

closed 2025-06-20 by github-actions[bot]